passinglink/passinglink #1: Documentation of DS4 key extraction

fluxcorenz, 2020-08-08:
Awesome project! I'm looking at setting it up (just received a knock-off blue pill and st-link to fiddle with) and will require instruction on how to extract the keys from my DS4 controllers for use with this.

jmgao, 2020-08-10:
Yeah, sorry, the drive I had all of the intermediates on isn't easily accessible to me (out of M.2 slots), so I'll have to reconstruct this from scratch (and it's been a year or so since I dumped the keys I've been using since).

Here's an outline while it's at the top of my mind so I don't forget anything (or if you're feeling adventurous):

hardware stuff:
 - solder wires to MD0, VCC, RST (and maybe GND. I just brushed the RST wire against one end of a capacitor that I probed to be GND)
 - boot to debug mode by connecting RST to GND with MD0 and VCC connected, boot to regular mode by connecting RST to GND without MD0 and VCC connected

finding untouched memory:
 - boot to debug mode
 - write 0xcc to all of RAM (I don't remember how I did this: I assume that using the usb flash programmer is going to explode as soon as you hit the stack or random global variables, but maybe they're located far enough down the address space that we have plenty of scratch space. I might have just written a loop in assembly, poked it into memory and jumped to it)
 - reboot to debug mode
 - read all of RAM to figure out which pages weren't touched during the reboot

copying flash to scratch pages:
 - write a trampoline to jump to an arbitrary address:
    IIRC, the usual tricks to load an immediate value didn't work because the FPB unit only works on code fetches, so it has to be something along the lines of `mov r0, 0xbeef; movt r0, 0xdead; bx r0`.
 - write a function that copies from flash to the scratch pages found above: there won't be enough space, so this will need to happen 5 or 6 times
 - set up the [FPB](https://developer.arm.com/documentation/ddi0337/h/debug/about-the-flash-patch-and-breakpoint-unit--fpb-) to replace the reset handler with the trampoline. the reset handler should be at the address contained in address 0x0.
 - boot to debug mode, set everything up, reboot to regular mode, wait a second, reboot to debug mode, fetch the saved data from the scratch pages, repeat N times

finding the key in flash:
  - boot normally, run [this code](https://github.com/jmgao/passinglink/blob/master/ds4dump/src/main.rs#L113) to do an authentication against the device, modify the code to save its signature
  - search for the signature ([key_sig](https://github.com/jmgao/passinglink/blob/master/ds4auth/src/lib.rs#L15)) in the dump: it'll be in a struct arranged like [this](https://github.com/jmgao/passinglink/blob/master/ds4auth/src/lib.rs#L141) (surrounded by pages of zeroes, iirc?)
  - convert the key to DER (i don't remember what I used to do this)
  - save the signature and serial to their own files

converting the key to a statically allocated array:
  - compile [this](https://github.com/passinglink/mbed_embed/blob/master/mbed_embed.cpp)
    - I don't remember how: IIRC libmbedtls-dev:i386 was close enough that it'll work
  - run it against the key, signature and serial
  - copy the generated header file to keys/, back it up, and never touch this again

dogtopus, 2020-11-18:
Well dumping the context object is a pretty horrible way of saving the private key IMO. Any reason why it was chosen instead of creating the context at runtime from private key/private CRT factors?

jmgao, 2020-11-18:
It's a horrible hack to save on RAM to be able to fit on bluepill.

dogtopus, 2020-11-18:
How much RAM would a context use though? I thought it would be more or less just a pointer table.

jmgao, 2020-11-18:
It's a pointer table to bigints that are dynamically allocated. Bloaty reports the following:

```
   0.1%     524 __limbs_ds4_key_RN                                    580   0.0%
   0.1%     260 __limbs_ds4_key_N                                     315   0.0%
   0.1%     256 __limbs_ds4_key_D                                     311   0.0%
   0.1%     256 __ds4_signature                                       309   0.0%
   0.1%     256 __ds4_key_e                                           305   0.0%
   0.1%     256 __ds4_key_n                                           305   0.0%
   0.0%     172 __ds4_key                                             218   0.0%
   0.0%     132 __limbs_ds4_key_P                                     187   0.0%
   0.0%     132 __limbs_ds4_key_Q                                     187   0.0%
   0.0%     128 __limbs_ds4_key_DP                                    184   0.0%
   0.0%     128 __limbs_ds4_key_DQ                                    184   0.0%
   0.0%     128 __limbs_ds4_key_QP                                    184   0.0%
   0.0%      16 __ds4_serial                                           66   0.0%
   0.0%       4 __limbs_ds4_key_E                                      59   0.0%
```

which sums up to 2648 bytes

dogtopus, 2020-11-18:
Hm that's pretty bad. It almost looks like it's copying all the stuff to RAM instead of just leave them as read-only. Like how tf is this embedded?

jmgao, 2020-11-18:
Eh, you more than that amount of memory to be able to do the bigint math needed for the crypto anyway (I trimmed it down with [yet another horrible hack](https://github.com/passinglink/mbedtls/commit/7e29449d7f4d929106124b1df98015865bb44e2e) that disables resistance to side-channel attacks), and the name is mbed**TLS**, after all, so they're probably expecting to be used on network capable parts with hundreds of kilobytes of memory.

I'm not super happy with it (well, okay, I'm a tiny bit gleeful), but I don't think this would get accepted upstream, and it doesn't seem likely to break.

Sidenote: I'm half-way done with a patch to move the keys out of the built image into a separate flash partition, so that if you have a key dump, manual rebuilding of master isn't needed.

dogtopus, 2020-11-18:
> Eh, you more than that amount of memory to be able to do the bigint math needed for the crypto anyway (I trimmed it down with [yet another horrible [...]

I gave up and just built a static mbed_embed on my RPi so that the ABI would match as much as possible and I could run it on both qemu or RPi.

jmgao, 2020-11-18:
I just pushed [a commit](https://github.com/passinglink/mbed_embed/commit/2ed6d4a6105966446d70d4c9290a641cb6e75cf1) that adds a CMakeLists.txt that compiles with the mbedtls included in zephyr. Back when I implemented this the first time around, I checked that the struct layout was identical on i386 and arm with `clang -cc1 -fdump-record-layouts`, so that should be fine.

dogtopus, 2020-11-18:
> I just pushed [a commit](https://github.com/passinglink/mbed_embed/commit/2ed6d4a6105966446d70d4c9290a641cb6e75cf1) that adds a CMakeLists.txt that compiles with the mbedtls included in zephyr. Back [...]

Well problem is my distro doesn't have 32-bit binaries anymore so it needs to be built from source.

jmgao, 2020-11-18:
> Also it would be nice to move all the bluepill haxxx to its own option(s) so that better boards won't get affected by those haxxx.

This means that I won't know if the hacks break until way later, potentially. I'd rather that dropping bluepill support be a conscious decision than "oops, it broke 6 months ago, and at this point, fuck it".

The hacks are arguably a strict improvement for the usecase, anyway: I'm not very concerned about someone using a side channel attack to extract my top secret encryption keys. :-)

dogtopus, 2020-11-18:
> This means that I won't know if the hacks break until way later, potentially. I'd rather that dropping bluepill support be a conscious decision than "oops, it broke 6 months ago, and at this point, [...]

Um... What about just testing it when there are major changes to mbedTLS for example?

The main problem is the current setup is still inherently fragile and letting bluepill haxxx go everywhere doesn't sound good to me.

jmgao, 2020-11-18:
> Um... What about just testing it when there are major changes to mbedTLS for example?

That's still more effort. (Realistically, I don't think I'm ever going to update mbedtls, since I'm not going to be using thread or whatever in zephyr that depends on it)

> The main problem is the current setup is still inherently fragile and letting bluepill haxxx go everywhere doesn't sound good to me. OTOH just let boards with enough resource use normal mbedTLS and create context from "DS4Key" in RAM (or even CCM) sounds much better.

I don't think I agree that this is fragile: it's definitely less fragile than having multiple code paths, one of which will be basically untested. Using the regular key parsing code is definitely going to cost more in flash and RAM, and I don't think it really buys anything, except initial setup (which should be much easier after moving it out to a separate partition, since I'll be forced to write a script to make that not a pain-in-the-ass.)

> BTW just verified that my static mbed_embed works nicely with a key I ripped from a certain type of bootleg controllers ages ago
Excellent :+1:

dogtopus, 2020-11-18:
Out of curiosity I checked gnuk's memory usage on FST-01, which is also using stm32f103 with the same RAM size, and...

```
   text	   data	    bss	    dec	    hex	filename
 109328	    240	  10336	 119904	  1d460	gnuk.elf
```

Comparing to PassingLink/Zephyr:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       42864 B        64 KB     65.41%
            SRAM:       20144 B        20 KB     98.36%
        IDT_LIST:         136 B         2 KB      6.64%
```
Well I guess Zephyr is just bloated :dizzy_face:

jmgao, 2020-11-18:
I don't think it's zephyr. Here's a bloaty dump: https://gist.github.com/jmgao/97b9967292557b416ebfcf2167552176

IIRC the allocator is massively oversized because I was experiencing mysterious crashes (which I'm pretty sure I've since figured out) so I increased it to be way bigger than needed to debug whether that was the problem.

jmgao, 2020-11-18:
(I'm a bit surprised by z_vprintk being there and having a pretty large static buffer, when logging is completely turned off, though.)

dogtopus, 2020-11-18:
Is the allocator just indicating what SRAM region can be `malloc()`ed?

Also

```
291   ---------- 0x2000378c LOCAL  OBJ    256      ps4::nonce
```

This can go away if you stream the nonce to sha256sum instead (of course that assumes mbedTLS allows streaming but most crypto libraries should have that)

jmgao, 2020-11-18:
Yeah, allocator is [this 10kB struct](https://github.com/passinglink/passinglink/blob/4f878683e92ff38e4a5f4417387ff374f98e2635/src/malloc.cpp#L107).

I'll take patches that optimize memory usage, but I don't think there's any point in spending much effort on it: it seems borderline impossible to be able to fit auth + UART on bluepill and everything else has tons of headroom.

dogtopus, 2020-11-18:
> Yeah, allocator is [this 10kB struct](https://github.com/passinglink/passinglink/blob/4f878683e92ff38e4a5f4417387ff374f98e2635/src/malloc.cpp#L107).
>
> I'll take patches that optimize memory usage, [...]

I thought UART is always a non-goal though. Maybe bare UART with reduced buffer (like the 64 byte arduino-style one for inter-chip communications) but I definitely won't even consider turning on shell or console or logging. That's purely asking for trouble.

dogtopus, 2020-11-24:
Just FYI: I added passinglink header support to my own tool (https://github.com/dogtopus/e3pp/blob/master/e3.py). It has no deps to mbedTLS and thus should be portable to every architecture including 64-bit ones. As long as you froze mbedTLS version (which you mentioned that you intend to do this) this should be a good alternative.

Support for partitions could come in 2 ways: either compile the header and generate the image using ldscript or natively support it given information about target ABI and offset.

Samebusa, 2022-09-02:
> Yeah, sorry, the drive I had all of the intermediates on isn't easily accessible to me (out of M.2 slots), so I'll have to reconstruct this from scratch (and it's been a year or so since I dumped [...]

I'm trying to follow this step-by-step and I get to boot it in debug mode but then I get stuck. Has anyone found any easier way to extract the key?

anonymousps4, 2023-10-23:
Here's my guide: https://github.com/anonymousps4/ds4blog

Should have all the steps including a Python tool.

fluxcorenz, 2023-10-23:
> Here's my guide: https://github.com/anonymousps4/ds4blog
> 
> Should have all the steps including a Python tool.

Beautiful writeup!