passinglink/passinglink #3: Not "fully" supported by the "Blue Pill" STM32F103 based board

bootsector, 2020-09-17:
I did some tests on the RSA signing part of mbedtls library using a STM32F103 with 20K of RAM (aka "Blue Pill") and here are the results:

1 - Signing takes about 7 seconds to complete: since the Blue Pill doesn't support async/threads, this will block inputs for that amount of time. This is slower than disconnecting USB and reconnecting it again after 8 minutes.

2 - In order to have visibility on memory consumption, I've used the stack based memory allocator (mbedtls_memory_buffer_alloc_init). That required 13K or RAM in order to have signing working. 7K remains for everything else, which seems too little for an RTOS, USB output logic for PS3, PS4, Switch and also user input polling.

So what I did was to use another Blue Pill as a "signing module" witn an I2C interface. So the main module (the one that takes care of input and USB) can asynchronously interface with it without adding too much lag/overhead.

Thanks for your work on this!

jmgao, 2020-09-17:
> 1 - Signing takes about 7 seconds to complete: since the Blue Pill doesn't support async/threads, this will block inputs for that amount of time. This is slower than disconnecting USB and reconnecting it again after 8 minutes.

No CPUs natively support threads: threads are implemented by the code running on them. (And every RTOS is going to have some sort of threading implementation: the "real-time" part of an RTOS is basically about predictability of thread scheduling).

The important part is that you can run the signing in a low-priority task that gets preempted by everything else, and tell the PS4 that you're not ready yet until it finally finishes. I set a timer to interrupt the task to poll user input and dispatch the USB output.

> 2 - In order to have visibility on memory consumption, I've used the stack based memory allocator (mbedtls_memory_buffer_alloc_init). That required 13K or RAM in order to have signing working. 7K [...]

I have signing working with 9768 bytes of dynamic memory allocation and 2304 bytes of stack, which lines up with your numbers. 7K is definitely enough: this project is an existence proof that it all fits. The application side of input/output (i.e. reading user input and converting it to a USB HID report) is essentially free when it comes to RAM: the cost is almost entirely in code size. If I'm counting right, it's something like 80 bytes of memory used total.

It's not a super fun development experience because you have to do stuff like completely turn off logging (UART is surprisingly expensive) to get things to fit, but it's definitely possible.

> So what I did was to use another Blue Pill as a "signing module" witn an I2C interface. So the main module (the one that takes care of input and USB) can asynchronously interface with it without adding too much lag/overhead.

I've been planning on coming up with a protocol for this and similar other tasks (e.g. hooking two MCUs together to use one to decode USB input and the other to translate it to PS4 or whatever). I've been waiting until I have an actually concrete use before nailing things down, though.

bootsector, 2020-09-17:
Spot on! 

Since I don't use any RTOS (I always do bare metal on STM32F103), I think that leveraging the signing portion to another module made all sense. And results have been great so far!

Such a module could even be used along with a "weaker" uC such as the Atmega32U4, that can do I2C, USB and input polling like a ninja! :)

Cheers!

dogtopus, 2020-11-10:
To free some CPU time you could also use passthrough/A7105 security chip or https://github.com/dogtopus/IllegalSecurityChip (I just finished writing it lol). The third one signs the challenge in under 0.5 seconds on a J3H145 card which also surprised me a bit when I saw it :P

closed 2020-11-18 by jmgao