passinglink/passinglink #4: DS4 auth passthrough?

dogtopus, 2020-11-10:
Since key extraction only works with specific hardware, it would be nice to have a passthrough setup instead

Having the ability to using the A7105 security chip would also be desireable since sometimes there might be cheap (new/used) PS4 licensed controller for sale.

jmgao, 2020-11-10:
Yeah, auth passthrough has been on the TODO list for a while and the PS5 being impending moves it up my priority list a bunch. I added Bluetooth support recently for an unrelated project (managing spectator mode in Gundam EXVS for streaming tournaments), and I'm planning on using that for tethered authentication to a smartphone (to either a controller plugged directly in or over the internet).

I've been wanting to make things more general, instead of being only an arcade stick, so another real user would be nice :-)
What would it take for you to switch from teensy to other hardware?

dogtopus, 2020-11-10:
Not a lot since I did have the plan to switch to STM32 (or at least make it an option) for better availability and higher performance per price ratio.

dogtopus, 2020-11-11:
Also I'm not sure supporting PS5 controller at this time would be the right thing to do. There seems to be some weird 64-bit checksum/MAC thingy going on with the USB input report and I have no clue what it is so far (even if it's just custom poly crc64 it would be practically impossible to just brute-force it because 64-bit). I bet that PS5 would check this as a way to "authenticate" the input report.

So my suggesstion now is to hold on, wait and see if more clue pops up that might uncover what it actually is. Things like poorly secured 3rd party controllers, hid-sony "accidentally" leak the algorithm or eventual PS5/DS5 firmware dump.

dogtopus, 2020-11-13:
After a bit of digging, it seems that Zephyr didn't have built in USB host support and the related [issue](https://github.com/zephyrproject-rtos/zephyr/issues/12386) went stale for almost 2 years. Is this why it ended up on your TODO list for so long?

jmgao, 2020-11-13:
Yeah, there are people actively working on it right now though.
From yesterday on [the zephyr slack](https://zephyrproject.slack.com/archives/CDUAL494Y/p1605119093112000?thread_ts=1605113928.110200&cid=CDUAL494Y): "Host support is WiP, a draft PR will be opened soon (delayed because of a second controller driver issues)."

dogtopus, 2020-11-13:
I'll start porting PL to my STM32F407 discovery board as soon as I got some time. I'll also make a simple host-only breakout board for the secondary USB (along with a lot of other boards I wanted to make for this year lol).