passinglink/passinglink #9: Rethink about Bluetooth support

dogtopus, 2020-11-18:
Seems that Zephyr does have HCI module support so finding a micro that does both USB and Bluetooth shouldn't be a priority from the get go. It seems to be a "standard" (at least was) for Bluetooth device designs to use an HCI module and a regular micro anyway.

The next question is: how incomplete is Zephyr's BR/EDR support and is it enough to at least get HID working?

dogtopus, 2020-12-06:
Just FYI: Seems that ESP-IDF does offer an "HCI mode" that allows the on-chip Bluetooth hardware to be used directly over UART as an HCI module, with [minimum coding](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart). This could be a cheap and easy way of sourcing Bluetooth Classic-capable modules (<$10 ESP32 modules/boards that are available pretty much everywhere)

dogtopus, 2021-08-02:
Some update on this: I managed to get the UART HCI passthrough example working with BlueZ+btattach. The max baud rate supported by IDF is a bit less than ideal so there's a bandwidth bottleneck but it shouldn't be that bad for DS4/Switch and there's also a patch (available)[https://github.com/espressif/esp-idf/pull/4844] that fixes it.

dogtopus, 2021-08-07:
> The next question is: how incomplete is Zephyr's BR/EDR support and is it enough to at least get HID working?

The answer seems to be yes. Looks like Zephyr always sucked at OOTB Bluetooth HID support even for BLE since there's no configurable "HID server" that takes a report descriptor plus some callbacks and it just works. This can be seen on the `peripheral_hids` example. Considering that Bluetooth Classic HID is just a thin (?) layer on top of L2CAP and Zephyr already got SDP implemented, it should be no less doable than BLE HID if nothing on the Zephyr side went terribly wrong.

henrebotha, 2022-03-31:
I'm confused. Isn't BT already implemented? https://github.com/passinglink/passinglink/blob/master/src/bt/bt.cpp

dogtopus, 2022-03-31:
> I'm confused. Isn't BT already implemented? https://github.com/passinglink/passinglink/blob/master/src/bt/bt.cpp

This is for BLE and input only, while the ticket focuses on Bluetooth Classic and output.

Also there is an issue currently regarding to the protocol reversing as I was unable to get the bluetooth controller of my choice to either talk to the PC (due to some BS they did to detect host device type) or sniffing the protocol (due to lack of equipment) so this is stalled indefinitely until I got another controller, ways to sniff the traffic or have more time to look into the full DS4 protocol (which is more complicated than 3rd party controllers)

jmgao, 2022-03-31:
> Ah, ok. Why does this need BT input? Is it to allow authenticating with a real DS4 without connecting it to the PassingLink controller with a wire?

I implemented it to use an nrf52840 dongle as a remote controller with inputs coming [from a PC](https://passinglink.github.io) mostly to help a friend do commentary on gundam tournaments, because the camera control while spectating in that game not great. My long term goal is to support passthrough of authentication to a controller connected to a phone (or to a computer across the internet), but I *still* haven't been able to get my hands on a PS5, so I haven't bothered with doing anything on that front.