passinglink/passinglink #11: Support multiple hosts and a way to switch between them?

sigboe, 2020-12-16:
I don't know if this is feasible or out scope.

I'd like to reduce the amount of controllers I have down to two, right now I have two PS4 controllers, 2 USB ps4 controllers, 2 8bit do controllers, 2 Switch pro controllers, 2 controllers for Android TV, and those are just the once I have within arms reach. Have more gaming devices not connected because they each need their own controllers.

I would love for a controller to have a feature to remember several hosts, maybe emulate different controller for each of the hosts. So I could have my controller connect to my PC, then I want to use my PS4 and switch over to my switch later. And before I go to bed I switch the same controller over to my [MiSTer](https://github.com/MiSTer-devel/Main_MiSTer/wiki).

My wife would be ever grateful to remove clutter form desks and drawers. And I would love to not have to charge a billion controllers

jmgao, 2020-12-17:
If you're okay with a wired-only arcade stick, this is doable now. That's probably not what you're looking for, though.

It's possible there there's an existing controller that has a microcontroller that passinglink could be ported to, and you could flash onto that, but there's probably a lot of work to support bluetooth classic for wireless PS4/switch support, and finding a controller is probably hard.

One alternative would be to get a bunch of [NRF52840 USB dongles](https://www.digikey.com/en/products/detail/nordic-semiconductor-asa/NRF52840-DONGLE/9491124), plug one of them into each console, and connect another controller to a computer that mirrors inputs to them selectively via Bluetooth LE. You'd add ~10ms of latency to inputs from this, but that's probably acceptable for most things.

dogtopus, 2021-01-02:
Sounds like you want some sort of converter that is cheap (so one can buy say 10 of them and plug into 10 separate devices) and works on multiple game consoles? I think PassingLink does have converter support on its roadmap but the problem is when.

> It's possible there there's an existing controller that has a microcontroller that passinglink could be ported to, and you could flash onto that, but there's probably a lot of work to support bluetooth classic for wireless PS4/switch support, and finding a controller is probably hard.

I was actually thinking about porting PassingLink onto some knockoff DS4 controllers that has Nuvoton M451 on board, only to find out that they now seem to be almost all RDA-based. My target has since then switched to making a JDM-05x-sized PCB that can be swapped into an authentic/sufficiently realistic knockoff DS4 and have a beefy enough micro that can be supported by PassingLink.