passinglink/passinglink #13: Toggle between dpad, mode_ls and mode_rs on pl_bluepill

piuzera, 2021-03-10:
Is there a way to do this currently? I've noticed that the stm32f4_disco has this function, but I can't get it to work on pl_bluepill

Thanks for this awesome project, by the way!

jmgao, 2021-03-12:
I implemented it [here](https://github.com/passinglink/passinglink/commit/bd907a14e2f09b455bc072f18f320d607408aef0), but it hasn't been merged into master yet because I had some rebasing to do, IIRC.

I need to rework the input system to make the giant nest of ifdefs for mode switches that may or may not be there cleaner, but it should be relatively straightforward to add a mode switch that controls the SOCD mode.

piuzera, 2021-03-12:
I shall wait for the merge to master then, since it may come with other features and whatnot.

Again, thanks for this! It has helped me quite a lot on prototyping some boards for myself.

closed 2021-03-12 by piuzera