passinglink/passinglink #14: Bluepill button output problems on v.1.1.1

piuzera, 2021-04-03:
Hello!

I don't know if any other person has gone through this problem, but it seems that the support for the Bluepill STM32F103C8 on the latest release is having issues. When building on the default specifications, the buttons stay ON when pressed and they also randomly stop working. Flashing the board also occasionally returns errors, skipping many pages.

I have tested this on two separate VMs and three separate bluepills, the problem persists.

I rolled back to the previous release and everything is working fine on all the boards. Just wanted to let you know about this issue, thanks again for this awesome project!

jmgao, 2021-04-04:
> When building on the default specifications, the buttons stay ON when pressed

This sounds like a bug in debouncing, can you try with commit 5caaa1906a034ae46257dc301503c080b81affbe and its parent?

> and they also randomly stop working.

If everything stops working when this happens (probably when you hit a direction?), this could be a stack overflow. Could you test 10dab75a24389f762252a0758f1e64d9aa04abbb to see if it still happens with that commit? If so, I'll probably need to do some refactoring to be able to turn off input profiles on bluepill.

> Flashing the board also occasionally returns errors, skipping many pages.

That's surprising: does it happen on all of your bluepills? The image is still well within flash limits, so unless you've got some bluepills with C6 MCUs, I don't know why this would be happening.

Xortes-Gimenez, 2021-04-09:
Hello, I tried to build the master using the debouncing commit [5caaa19](https://github.com/passinglink/passinglink/commit/5caaa1906a034ae46257dc301503c080b81affbe) but this error appears many times:

error: macro "PL_GPIO" passed 3 arguments, but takes just 2

Thanks for the amazing project, great job!

rlcardoo, 2021-07-13:
```
> Hello!
> 
> I don't know if any other person has gone through this problem, but it seems that the support for the Bluepill.....
```
Hello Friend.
I am also having problems with the BLUE Pill STM32F103c8t6.  I have to disconnect and connect many times to be able to recognize it as a joystick, and when I do, when I press a button it stays on. 
Could you send me the previous Bin that works perfectly?  My email: ricardo.ams@gmail.com
 Thank you very much.

x0rtes, 2021-08-09:
Hello, I saw the recent changes into master, but I still have this issue, is there any way to solve it? @jmgao @dogtopus

Thanks!

dogtopus, 2021-08-09:
I am able to reproduce this. For me it's even worse: the buttons are either not working or are stuck along with some other buttons. Seems like something definitely went very wrong.

x0rtes, 2021-08-19:
@dogtopus It was needed for me to change  the bluepill_pins.dtsi to have some response at all, even on v1.1.0 (which dont have this bug once .dtsi have changed buttons)... so I'm still on v1.1.0 I would really like to go to v1.1.1 since it has socd last input support.

rlcardoo, 2021-08-25:
@x0rtes  Please send me version v1.1.0 I am unable to compile and the bin file is no longer available. Thank you very much. my email: ricardo.ams@gmail.com

Bagre-de-Chapeu, 2022-09-28:
Can someone send me version 1.0 for the bluepill too?
My email is fredscmorais@gmail.com