passinglink/passinglink #15: Provisioning script with wrong path?

x0rtes, 2021-04-27:
I inserted ds4.der ds4.serial ds4.sig into keys folder and changed ds4.inhl.h HAVE_DS4_KEY 1
Trying on Bluepill...

> ~/passinglink/passinglink# ./scripts/provision.sh "arbitrary name string" ds4.der ds4.serial ds4.sig
> MCUboot not supported on board 'pl_bluepill'
> + DTS_HEADER=build/pl_bluepill/pl/zephyr/include/generated/devicetree_unfixed.h
> + [[ ! -f build/pl_bluepill/pl/zephyr/include/generated/devicetree_unfixed.h ]]
> + HEADER='#include <devicetree.h>\n'
> + HEADER='#include <devicetree.h>\n\n#define addr DT_REG_ADDR(DT_NODELABEL(provisioning_partition))'
> + HEADER='#include <devicetree.h>\n\n#define addr DT_REG_ADDR(DT_NODELABEL(provisioning_partition))\n#define size DT_REG_SIZE(DT_NODELABEL(provisioning_partition))'
> + HEADER='#include <devicetree.h>\n\n#define addr DT_REG_ADDR(DT_NODELABEL(provisioning_partition))\n#define size DT_REG_SIZE(DT_NODELABEL(provisioning_partition))\naddr size'
> + PARTS=($(
>   echo -e "$HEADER" |
>   cpp -I "$BUILD_DIR/pl/zephyr/include/generated" -I "$ROOT/zephyr/include" -w - |
>   tail -n 1
> ))
> ++ echo -e '#include <devicetree.h>\n\n#define addr DT_REG_ADDR(DT_NODELABEL(provisioning_partition))\n#define size DT_REG_SIZE(DT_NODELABEL(provisioning_partition))\naddr size'
> ++ tail -n 1
> ++ cpp -I build/pl_bluepill/pl/zephyr/include/generated -I /root/passinglink/passinglink/scripts/../../zephyr/include -w -
> + OFFSET=61440
> + SIZE=4096
> + pyocd list -t
> + egrep -q '^\s+stm32f103c8\s'
> + /root/passinglink/passinglink/scripts/../../tools/mbed_embed/mbed_embed 61440 4096 'arbitrary name string' ds4.der ds4.serial ds4.sig
> mbed_embed: failed to open '61440': No such file or directory

Since tools is a folder on the first passinglink folder, I think it will not be able to acess this way. Maybe I'm doing something wrong.

Thanks in advance!

jmgao, 2021-04-27:
Whoops, the error message there is wrong: I [forgot to update the argv offsets](https://github.com/passinglink/mbed_embed/blob/b66f57cf79eddcc7a0c6849532c2e07a9610a60c/mbed_embed.cpp#L150) when adding new arguments.

My guess would be that you're passing the wrong paths to the files, you probably want `./scripts/provision.sh "arbitrary name string" keys/ds4.der keys/ds4.serial keys/ds4.sig`.

jmgao, 2021-04-27:
(If you run `west update`, it should fetch an updated version of mbed_embed that has fixed error messages)

x0rtes, 2021-04-27:
@jmgao thanks I will try and report

x0rtes, 2021-04-27:
Tested after west update and using keys/ before each key... new error ValueError

> ~/passinglink/passinglink# ./scripts/provision.sh "arbitrary name string" keys/ds4.der keys/ds4.serial keys/ds4.sig
> MCUboot not supported on board 'pl_bluepill'
> + DTS_HEADER=build/pl_bluepill/pl/zephyr/include/generated/devicetree_unfixed.h
> + [[ ! -f build/pl_bluepill/pl/zephyr/include/generated/devicetree_unfixed.h ]]
> + HEADER='#include <devicetree.h>\n'
> + HEADER='#include <devicetree.h>\n\n#define addr DT_REG_ADDR(DT_NODELABEL(provisioning_partition))'
> + HEADER='#include <devicetree.h>\n\n#define addr DT_REG_ADDR(DT_NODELABEL(provisioning_partition))\n#define size DT_REG_SIZE(DT_NODELABEL(provisioning_partition))'
> + HEADER='#include <devicetree.h>\n\n#define addr DT_REG_ADDR(DT_NODELABEL(provisioning_partition))\n#define size DT_REG_SIZE(DT_NODELABEL(provisioning_partition))\naddr size'
> + PARTS=($(
>   echo -e "$HEADER" |
>   cpp -I "$BUILD_DIR/pl/zephyr/include/generated" -I "$ROOT/zephyr/include" -w - |
>   tail -n 1
> ))
> ++ echo -e '#include <devicetree.h>\n\n#define addr DT_REG_ADDR(DT_NODELABEL(provisioning_partition))\n#define size DT_REG_SIZE(DT_NODELABEL(provisioning_partition))\naddr size'
> ++ cpp -I build/pl_bluepill/pl/zephyr/include/generated -I /root/passinglink/passinglink/scripts/../../zephyr/include -w -
> ++ tail -n 1
> + OFFSET=61440
> + SIZE=4096
> + egrep -q '^\s+stm32f103c8\s'
> + pyocd list -t
> + /root/passinglink/passinglink/scripts/../../tools/mbed_embed/mbed_embed 61440 4096 'arbitrary name string' keys/ds4.der keys/ds4.serial keys/ds4.sig
> + pyocd flash -e sector -a 61440 -t stm32f103c8 build/pl_bluepill/provisioning.bin
> 0002886:CRITICAL:__main__:no memory region defined for address 0x0000f000
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.8/dist-packages/pyocd/__main__.py", line 402, in run
>     self._COMMANDS[self._args.cmd](self)
>   File "/usr/local/lib/python3.8/dist-packages/pyocd/__main__.py", line 576, in do_flash
>     programmer.program(self._args.file,
>   File "/usr/local/lib/python3.8/dist-packages/pyocd/flash/file_programmer.py", line 156, in program
>     self._format_handlers[file_format](file_obj, **kwargs)
>   File "/usr/local/lib/python3.8/dist-packages/pyocd/flash/file_programmer.py", line 175, in _program_bin
>     self._loader.add_data(address, data)
>   File "/usr/local/lib/python3.8/dist-packages/pyocd/flash/loader.py", line 115, in add_data
>     raise ValueError("no memory region defined for address 0x%08x" % address)
> ValueError: no memory region defined for address 0x0000f000
>

jmgao, 2021-04-30:
> pyocd flash -e sector -a 61440 -t stm32f103c8 build/pl_bluepill/provisioning.bin
> 0002886:CRITICAL:main:no memory region defined for address 0x0000f000

My guess is that this is expecting an address in the flash memory area starting from `0x08000000`. You could try running `pyocd flash -e sector -a 134279168 -t stm32f103c8 build/pl_bluepill/provisioning.bin`, but I'm not certain if the mapping at address 0 points into flash on zephyr.

My bluepills are buried in a drawer somewhere, I'll try to find them and take a look at this (and the broken inputs) over the weekend.

x0rtes, 2021-05-05:
@jmgao I tried

> pyocd flash -e sector -a 134279168 -t stm32f103c8 build/pl_bluepill/provisioning.bin

As you said and the flash occurs normally, but didn't work on the console. I don't know if it's related to the flashing or if the keys that I have are not working.

closed 2021-05-10 by x0rtes