passinglink/passinglink #16: Is there a second auth scheme?

itewqq, 2021-07-09:
Hello, just discovered your great work here!

But when I researched the nxp A710x series in my ps4 controller, I found that it has another auth mode besides providing a response to the input challenge of 0x100 length. In fact, when I input 0x01 to the chip, it will directly return an ascii code to tell me that it is now in auth2 mode：

```
# Input 0x01
uart_event_handler()
-------- uart_rxcnt = 1
AUTH2 command!
```

It seems to provide keep-alive verification for the wireless connection of the handle. Both input and output are 0x20 bytes.

Do you have any research on this verification mechanism? Or there are any guiding directions to study it, such as how you learned about these algorithms. I would be very grateful if you can provide some help!

Finally, thank you again for the great work in this repo.

jmgao, 2021-07-09:
AFAIK, authentication over bluetooth is essentially the same as USB. Here are some resources on the [BT](https://www.psdevwiki.com/ps4/DS4-BT) and [USB](https://www.psdevwiki.com/ps4/DS4-USB) protocols.

Maybe this is related to device pairing? You could try inspecting the bytes to see if they look like the device's bluetooth address.

itewqq, 2021-07-09:
@jmgao Thanks for the quick reply!!! Going to check the bl protocol out, thank you very much!

dogtopus, 2021-07-09:
Actually there seems to be one. There's SET_REPORT 0x03 and GET_REPORT 0x04 which are Bluetooth-only. Failing them seems to cause the PS4 to immediately ignore the controller.

closed 2021-07-10 by itewqq