passinglink/passinglink #17: RP2040 support possible/probable?

sigboe, 2021-07-19:
Just wondering if a port to the rp2040 is possible or probable. From just looking at the specs it should be possible right?
It doesn't have internal storage, but supports external storage and usually ships on boards with example 16MB.

Its also dual core 32 bit, maybe even PS5 would be possible in the future? I've seen a Chinese vendor (Brook Accessories) making ready to [ship this August a co processor for one of their arcade pcb which will add PS5 support.](https://pbs.twimg.com/media/E5-fLakVoAIbRBd?format=jpg&name=medium) Would be interesting to see how that works

jmgao, 2021-07-20:
There are some introductory efforts to port zephyr to RP2040 (https://github.com/zephyrproject-rtos/zephyr/pull/33172, https://github.com/zephyrproject-rtos/zephyr/pull/34835), once that happens, everything should Just Work after updating the zephyr version used for passinglink.

> I've seen a Chinese vendor (Brook Accessories) making ready to ship this August a co processor for one of their arcade pcb which will add PS5 support.

It's unclear to me whether this is DRM for an upsell, it's actually needed for performance reasons, or if it's actually a legitimate security chip harvested from a licensed controller.

sigboe, 2021-07-20:
> It's unclear to me whether this is DRM for an upsell, it's actually needed for performance reasons, or if it's actually a legitimate security chip harvested from a licensed controller. 
@jmgao 
The company has said in social media that they tried to do it just as a firmware upgrade, but this was the solution they had to. I don't know if this is true. But I presume it is possible in china buying grey market chips flashed by a company that has a license to make licensed controllers.

dogtopus, 2021-08-02:
> Its also dual core 32 bit, maybe even PS5 would be possible in the future? I've seen a Chinese vendor (Brook Accessories) making ready to [ship this August a co processor for one of their arcade pcb [...]

I'd like to look into this more. However I don't have PS5 and the security chip is going to be a blackbox so no more test scripts for it.

Nicknakin, 2022-02-06:
> There are some introductory efforts to port zephyr to RP2040 ([zephyrproject-rtos/zephyr#33172](https://github.com/zephyrproject-rtos/zephyr/pull/33172), [zephyrproject-rtos/zephyr#34835](https://github.com/zephyrproject-rtos/zephyr/pull/34835)), once that happens (and peripheral support is implemented), everything should Just Work after updating the zephyr version used for passinglink.
> 
> > I've seen a Chinese vendor (Brook Accessories) making ready to ship this August a co processor for one of their arcade pcb which will add PS5 support.
> 
> It's unclear to me whether this is DRM for an upsell, it's actually needed for performance reasons, or if it's actually a legitimate security chip harvested from a licensed controller.

Just wanted to drop a comment to mention that it appears that there has been substantial progress on these issues as both have been merged and closed, not sure if it's enough for this project to begin supporting the 2040 but worth checking out!

Nicknakin, 2022-02-08:
https://docs.zephyrproject.org/latest/boards/arm/rpi_pico/doc/index.html
Official support page on zephyr docs for the RP2040

jmgao, 2022-02-15:
I think the zephyr support is still not quite good enough (e.g. no USB support), but it looks like it's going to be usable soon. I've got some RP2040s coming in the mail!