passinglink/passinglink #18: Need more information about how it works

mihannnik, 2021-07-29:
The repository lacks information on how the device should work and be configured, as well as examples of use and configuration.
I just cannot compile this because I do not have a machine with UNIX system and my understanding of microcontroller code is low.
Can you add more examples in README and default releases in releases tab (not only source code)?
Or just more information how to configure it

dogtopus, 2021-08-02:
You could run a Ubuntu VM or WSL2 (if you use Windows) and follow the [instruction](https://github.com/passinglink/passinglink#compiling).

mihannnik, 2021-08-02:
Ok, i can compile it, but i don't know how to configure it for me. Where i can find config file with buttons setup?

dogtopus, 2021-08-02:
Search GPIO after you open `menuconfig`. There should be an option to enable GPIO input.

Check the overlay DTS files under `boards/` to know what pins are used for what buttons. (e.g. the bluepill one is [here](https://github.com/passinglink/passinglink/blob/master/boards/bluepill_pins.dtsi))

mihannnik, 2021-08-11:
When i try to compile project i get error:
```
pc@pc-VirtualBox:~/passinglink/passinglink$ ./scripts/build.sh
MCUboot supported on board 'stm32f4_disco'
+ mkdir -p build/stm32f4_disco
+ [[ 1 == 1 ]]
+ [[ '' != 1 ]]
++ realpath /home/pc/passinglink/passinglink/scripts/../../passinglink/boards/stm32f4_disco.overlay
+ OVERLAY_FILE=/home/pc/passinglink/passinglink/boards/stm32f4_disco.overlay
+ OVERLAY=
+ [[ -f /home/pc/passinglink/passinglink/boards/stm32f4_disco.overlay ]]
+ OVERLAY=-DDTC_OVERLAY_FILE=/home/pc/passinglink/passinglink/boards/stm32f4_disco.overlay
+ '[' '!' -d build/stm32f4_disco/mcuboot ']'
+ west build -d build/stm32f4_disco/mcuboot
[2/237] Generating zephyr/autogen-pubkey.c
FAILED: zephyr/autogen-pubkey.c 
cd /home/pc/passinglink/passinglink/build/stm32f4_disco/mcuboot && /usr/bin/python3.8 /home/pc/passinglink/bootloader/mcuboot/scripts/imgtool.py getpub -k /home/pc/passinglink/bootloader/mcuboot/root-rsa-2048.pem > /home/pc/passinglink/passinglink/build/stm32f4_disco/mcuboot/zephyr/autogen-pubkey.c
Traceback (most recent call last):
  File "/home/pc/passinglink/bootloader/mcuboot/scripts/imgtool.py", line 19, in <module>
    from imgtool import main
  File "/home/pc/passinglink/bootloader/mcuboot/scripts/imgtool/main.py", line 25, in <module>
    from imgtool import image, imgtool_version
  File "/home/pc/passinglink/bootloader/mcuboot/scripts/imgtool/image.py", line 24, in <module>
    from .boot_record import create_sw_component_data
  File "/home/pc/passinglink/bootloader/mcuboot/scripts/imgtool/boot_record.py", line 19, in <module>
    import cbor
ModuleNotFoundError: No module named 'cbor'
[3/237] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/printk.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/pc/passinglink/passinglink/build/stm32f4_disco/mcuboot
```
Any ideas?

dogtopus, 2021-08-11:
For stm32f4_disco more python dependencies are required. I think they are `cbor`, `cryptography`, `click` and `intelhex`. Just install them with pip in the same python environment used by west.

mihannnik, 2021-08-18:
Ok, i compile it and when try to flash i get error: only STM32F407 support in script. My board is F411, can it work on it? When i tried flash using STM32 Cube Downloader it flashed normaly, but when connecting to PC it not defined as controller.