passinglink/passinglink #27: Possible Xbox 360 console support?

IcicaSora, 2022-01-15:
Since there really isn't any cheap(er) X360 fightboard alternatives and there exist some information on the Internet about the auth process between the controller and the console, wonder if adding xbox 360 support is doable (or viable on something like the bluepill).

dogtopus, 2022-03-27:
No, because the pre-shared (?) secret for xbox360 controller auth is still not public so passthrough seems to be the way to go. However Zephyr still doesn't have USB host support and third party, portable, open source USB host stack is quite an unobtainium in practice.

aenertia, 2022-11-20:
TinyUSB is quite usable at this point and used in several similar projects https://github.com/hathach/tinyusb

Check out i.e : https://github.com/sanjay900/Ardwiino