passinglink/passinglink #30: Flashing Adafruit NRF52840 Express, Board Support

JohnPeng47, 2022-10-02:
Hi I am trying to flash the Adafruit NRF52840 Express board, but running into issues with PyOCD not being able to detect my board. Specifically, I think my issue is putting the board into DFU mode to accept the debug probes. To be honest, I am not sure if this is a problem on my end, or it a problem with Passinglink, so I apologize if this is an issue with me not properly setting the board into DFU mode (holding down the USERSWTCH after RESET). While looking at documentation online, I saw that the GPIO pin 7 is mapped to a switch putting it into DFU mode:

`User/DFU Switch

A tactile switch is provided for use in your projects, which is connected to P1.02 and is accessible in code as D7 in Arduino and SWITCH in CircuitPython.

Holding this button down coming out of a board reset will also force the device to enter and remain in USB bootloader mode, which can be useful if you lock your board up with bad application code!`

However, in the passinglink scripts/boards.sh file, I did not see any BOOT_USB_DFU_DETECT_PIN mapped to pin 7. 

`elif [[ "$BOARD" == "pl_dongle" ]]; then
  PL_PYOCD_TYPE=nrf52840

  PL_MCUBOOT_SUPPORTED=1
  PL_MCUBOOT_OPTS="
    -DCONFIG_I2C=y
    -DCONFIG_PASSINGLINK_DISPLAY_SSD1306=y
    -DCONFIG_BOOT_USB_DFU_GPIO=y
    -DCONFIG_BOOT_USB_DFU_DETECT_PORT=\"GPIO_0\"
    -DCONFIG_BOOT_USB_DFU_DETECT_PIN=18
  "
`
So my question is, is flashing Adafruit NRF52840 Express directly supported in Passinglink or do I need to define my own board configuration for it?

jmgao, 2022-10-02:
Are you using a debug probe connected to the SWD connector? DFU mode is for updating over USB, with an already-flashed bootloader (passinglink has its own bootloader that replaces the one that comes preflashed on your board), so if you have the board connected to a debug probe and it's not showing up in pyocd, something else is wrong.

(You'll probably want to define a new configuration eventually anyway, the pin definitions probably collides with some of the onboard hardware like the LED or the QSPI flash chip).

JohnPeng47, 2022-10-03:
No, I'm just using the plain ol USB connector, don't have anything hooked
up to SWD. So you're saying even without putting the board int DFU mode,
there should be debug probe sent across the USB connection?

jmgao, 2022-10-03:
No. You need an SWD programmer to flash passinglink the first time.

closed 2022-10-03 by jmgao