passinglink/passinglink #35: PS5 Support

JStyle21, 2025-05-23:
Hi,

Can the same dumping method that worked for the dualshock4  be used with the dualsense or have they changed something?
Also if a ps4 passthrough works on a PS5 why can't it be used with the extracted keys from a dualshock4?

JStyle21, 2025-05-23:
Adding this as well
https://gist.github.com/dogtopus/894da226d73afb3bdd195df41b3a26aa
https://github.com/sensepost/dual-pod-shock/blob/master/dualsense.c

xromansx, 2025-11-18:
I don't think they emulate it, maybe they simply have it or in sort of white list I don't know for sure but not all fighting boards pass authentication. or maybe there is used ps4 authentication and whitelisted id.

xromansx, 2025-11-20:
As I know fighting sticks that ps5 accepts where originally ps4 one.  Joysticks it only accepts in compatibility mode. Maybe I'm wrong.

JStyle21, 2025-11-21:
Actually i can answer myself after some digging

PS4
Early DS4 controllers stored the private key in a normal MCU, so the key could be pulled out with hardware tricks. Later PS4 hardware used real secure chips, but passthrough still worked because the PS4 did not check as strictly.

PS5
DualSense always uses a real secure element and the private key cannot be dumped or copied. Because of that, PS5 accessories only work in three ways

Licensed products use Sony supplied authentication chips

Unlicensed adapters rely on passthrough to a real controller which is why they time out

Some small mod projects harvest a real DualSense chip

No one is cloning the PS5 security chip and no one is extracting its keys.